moveit/moveit_core #235: Possible bug in transformVector3

dornhege, 2015-03-16:
Hi,
I just stumbled upon this, so if this is actually correct please disregard it.

    void transformVector3(const std::string &from_frame, const Eigen::Vector3d &v_in, Eigen::Vector3d &v_out) const
    {
        v_out = getTransform(from_frame).rotation() * v_in;
    }

The code only takes the rotation part. I believe it should use the complete transform to transform a Vector3.

sachinchitta, 2015-03-16:
transformVector3 seems to assume that v_in and v_out are "free" vectors, not points. This should be in the documentation though - it looks like its not used anywhere else within moveit_core or moveit_ros.

dornhege, 2015-03-16:
OK, makes sense. In that case I'd just leave this open as a "documentation bug".

davetcoleman, 2015-03-17:
Why not just improve the documentation? https://github.com/ros-planning/moveit_core/pull/236

closed 2015-03-17 by davetcoleman