moveit/moveit_core #239: capsule support in moveit_core

arocchi, 2015-04-28:
Hello everyone,
does moveit support as of now capsule geometries when checking collisions?
The capsule geometry distance computation has been implemented last year in fcl (see https://github.com/flexible-collision-library/fcl/pull/37) and it would be nice to access this feature directly through MoveIT (see https://github.com/Karsten1987/sot_fcl_distance_computation/blob/master/src/DistanceComputation.cpp where a cylinder geometry is treated as a capsule)

mikeferguson, 2015-05-06:
AFAIK, all of the moveit collision checking is limited to types that can be represented by a shape_msgs message -- of which I believe there is no capsule.

arocchi, 2015-05-06:
Yes you're right https://github.com/ros/common_msgs/blob/jade-devel/shape_msgs/msg/SolidPrimitive.msg it's a problem of the whole stack... that should e addressed starting from the ros/common_msgs repo :\

davetcoleman, 2015-05-06:
Doesn't seem like it would be hard to add a capsule solid primitive

arocchi, 2015-05-07:
@davetcoleman does it mean you are willing to take a stab at it? :)

davetcoleman, 2015-05-07:
Sorry, I have no need for capsule geometries at the moment

arocchi, 2015-05-08:
@davetcoleman if I have time in the future I might take a look at it. Anyway, when I was talking about the whole stack, I intended things like urdfdom :\

mikeferguson, 2016-08-25:
ported to moveit repo, see linked issue above

closed 2016-08-25 by mikeferguson