moveit/moveit_core #240: class fcl::CollisionObject’ has no member named collisionGeometry

StanleyYake, 2015-05-07:
Hi there:
    Because I followed the "Ros by example volume 2" writed by R.Patrick Goebel, he suggest to build it from source.But I encountered with this after catkin_make in my catkin_ws/src directory: class fcl::CollisionObject’ has no member named ‘collisionGeometry'
[image]

mikeferguson, 2015-05-07:
If there is going to be any chance of getting help on this, this ticket is going to need a lot more information -- what ROS distro are you using? what OS version? what branch/tag have you checked out for each repo that you are building from source? what version of FCL are you using, and is it from debians or source?

StanleyYake, 2015-05-08:
Hi mikeferguson:
I 'm using the ROS Hydro on Ubuntu 12.04, and used the default repo when building from source, thx for you reminds, the mistake is that the default repo is indigo, after changing to Hydro, I have build it successfully, thx again :+1:

closed 2015-05-08 by StanleyYake