moveit/moveit_core #249: Dynamically Reconfigure Robot/Scene

ndantam, 2015-06-02:
It would be useful to able to dynamically change the kinematics of the
robot.  For example, picking a tray containing several objects should
incorporate the child objects of the tray into the robot.  This is
roughly possible by attaching as collision objects all children of the
picked object; however, this is an additional step for something which
could be managed based on the kinematic tree of the robot and scene.

It seems that this may be difficult to implement since moveit
separates the RobotModel and the PlanningScene.  Are the underlying
data structures similar?  Does the PlanningScene consider
relative transforms or are all object poses in the global frame?

I think the simplest implementation would be to treat both robot and scene as
kinematic trees using the same data structure.

dornhege, 2016-08-01:
MoveIt does not have a kinematic tree of the scene. All objects are considered to be in the planning frame (unless attached). I'm not sure, if the added complexity of attached fixed kinematic chains instead of objects warrants this feature. These will not be treated any different during planning as all objects being directly attached.

mikeferguson, 2016-08-25:
This is a fairly extensive feature request that has seen no movement in a while, closing as we are porting this repo -- if this is still an issue, please open a new ticket against https://github.com/ros-planning/moveit

closed 2016-08-25 by mikeferguson