moveit/moveit_core #260: recent merge #259 breaks indigo build

rhaschke, 2015-10-20:
The recent merge of PR #259 performed by @davetcoleman breaks indigo build:
The addition of the new KinematicsQueryOption `KinematicsQueryOptions` in [`kinematics_base.h`](https://github.com/ros-planning/moveit_core/blob/8759d5125a1a2e4d1151f3673bb19124fead2bc6/kinematics_base/include/moveit/kinematics_base/kinematics_base.h#L116) is not mirrored by a change in [`moveit_ros/robot_interaction/.../kinematic_options.cpp`](https://github.com/ros-planning/moveit_ros/blob/indigo-devel/robot_interaction/src/kinematic_options.cpp#L92-L93) causing the [static assert to fail](https://github.com/ros-planning/moveit_ros/blob/indigo-devel/robot_interaction/src/kinematic_options.cpp#L118-L121). The ` jade-devel` branch has the appropriate changes.

shaun-edwards, 2015-10-20:
@jrgnicho, can you verify and fix this issue?

shaun-edwards, 2015-10-20:
There is still an open PR that needs to be merged to fix this: ros-planning/moveit_ros#617.  It was opened along with the PR #259, but it wasn't merged.  @davetcoleman, can you review and merge?

davetcoleman, 2015-10-20:
Not sure if we should merge #617 or revert this - @sachinchitta wanted to review #259 for indigo

davetcoleman, 2015-10-22:
it has been merged https://github.com/ros-planning/moveit_core/pull/259

closed 2015-10-22 by davetcoleman