moveit/moveit_core #270: Next MoveIt! Release

davetcoleman, 2015-12-10:
Looks like we haven't had a Indigo MoveIt! release since last January, I'd suggest we do one soon before a whole year passes by. Thoughts?

@mikeferguson

shaun-edwards, 2015-12-10:
There are some changes that we submitted to support [Descartes](https://github.com/ros-industrial-consortium/descartes) that we would really like to see released.

mikeferguson, 2015-12-10:
I agree -- we should do a release -- let me talk with @tfoote about the sync schedule over the next few weeks as the last sync was about two weeks and we will want to make sure when we do the release we have significant soak time to test the shadow-fixed debians.

ShawnSchaerer, 2015-12-14:
+1 on the Descartes integration.  This is something that we are actively working on so having it under one software interface would be ideal

davetcoleman, 2015-12-14:
@shaun-edwards could you make a checklist here of what is left for Descartes?

davetcoleman, 2016-01-06:
I'd really like to get this release out now. I just emailed Tully asking about the schedule.

shaun-edwards, 2016-01-06:
@davetcoleman, sorry I missed your comment.  The Descartes changes have been committed to Indigo.  The only missing piece is the FastIK generator, but I don't believe we want to port those changes back to Indigo (they are in the jade branch).

davetcoleman, 2016-01-06:
@tfoote on the next release:
> We're close right now. Though it looks like we might have a few regressions at the moment. Once we resolve those I'll plan to sync. 
We'll release MoveIt! right after that.

130s, 2016-01-27:
Ping ping thanks thanks!

davetcoleman, 2016-06-24:
indigo was released in May https://github.com/ros-planning/moveit_core/releases/tag/0.8.0, closing

closed 2016-06-24 by davetcoleman