moveit/moveit_core #279: "Returning dirty link transformation" warning when planning with joint constraints

johnnieleung, 2016-02-25:
Problem: when planning with joint constraints, warning message "Returning dirty link transformation" keeps popping up.

Problem location: I have tracked the problem down to the following function in moveit_core/constraint_samplers/src/default_constraint_samplers.cpp:

```
bool constraint_samplers::JointConstraintSampler::sample(robot_state::RobotState &state,
                                                         const robot_state::RobotState & /* reference_state */,
                                                         unsigned int /* max_attempts */)
{
  ...
  state.setJointGroupPositions(jmg_, values_);
  return true;
}
```
The problem is that after `state.setJointGroupPositions(jmg_, values_)` the state is not updated. Therefore when followed by a call to an IKConstraintSampler, the robot state is actually dirty, thus causing the warning message.

Fix: Just by adding `state.update()` after the `state.setJointGroupPositions(jmg_, values_)` in above code solves the problem.

DAIChengkai, 2016-02-25:
Hi, I also have the same issue you mentioned in the #278 , could you told me the way you solve it? Thanks!

johnnieleung, 2016-02-25:
Hi, please check out the reply I just added in #278

closed 2016-08-24 by mikeferguson