moveit/moveit_core #283: malloc() memory corruption

ibaranov-cp, 2016-04-14:
A client is running MoveIt!, and having malloc memory corruption errors. At first, I suspected it to be a config problem, but after remotely connecting to their machine, I see the issue on running the setup_assistant, and loading our basic Husky description. I have used both Husky description and moveit assistant many times and have never had this issue.

```
[ INFO] [1460566598.776610181]: Robot semantic model successfully loaded.
[ INFO] [1460566598.776628693]: Setting Param Server with Robot Semantic Description
[ INFO] [1460566598.794987563]: Loading robot model 'husky'...
*** Error in `/opt/ros/indigo/lib/moveit_setup_assistant/moveit_setup_assistant': malloc(): memory corruption: 0x0000000002134820 ***
```

Similar issues arise when using an already tested and working moveit config (husky + UR5)
```
[INFO] [WallTime: 1460565535.211129] Controller Spawner: Waiting for service controller_manager/load_controller
*** Error in `/opt/ros/indigo/lib/moveit_ros_move_group/move_group': malloc(): memory corruption: 0x0000000001a63e30 ***
[move_group-3] process has died [pid 8191, exit code -6, cmd /opt/ros/indigo/lib/moveit_ros_move_group/move_group arm_controller/follow_joint_trajectory:=follow_joint_trajectory __name:=move_group __log:=/home/administrator/.ros/log/e59e2c54-0194-11e6-b9eb-fcaa14dddb25/move_group-3.log].
log file: /home/administrator/.ros/log/e59e2c54-0194-11e6-b9eb-fcaa14dddb25/move_group-3*.log
```

There is no log file generated (above file does not exist), I am starting to re-compile with gdb support enabled, and taking a deeper look. (apt repos are all updated, both source and debs versions exhibit the same issue)

If you have any idea as to possible causes, this would help greatly.

TheDash, 2016-05-05:
@davetcoleman Thoughts?

CMobley7, 2016-06-09:
Any luck solving this problem?

CMobley7, 2016-06-09:
@TheDash It should be working on our robot; however, I was making a new srdf on my computer with links I've added. Following the advice on another thread, I used netfabb to repair and re-export all of the stl. After which, the problem was fix. Very few of the stl didn't need to be repaired on export.

TheDash, 2016-06-09:
Good to hear. Can you confirm (for my knowledge and others sake) that the issue is bad STLs?

CMobley7, 2016-06-09:
Yes, the only things I changed between runs of the Moveit Setup Assistant was the STLs?

TheDash, 2016-06-09:
Awesome! thank you sir.

closed 2016-06-14 by davetcoleman