moveit/moveit_core #309: New Continuous Integration Method

davetcoleman, 2016-07-08:
I've created yet another continuous integration script - its a combination of the strengths of [industrial_ci](https://github.com/ros-industrial/industrial_ci) and the simplicity of @jbohren and my [original MoveIt! travis](https://github.com/ros-planning/moveit_core/blob/09bbc196dd4388ac8d81171620c239673b624cc4/.travis.yml) script that was recently revitalized by @rhaschke. The show-stopper problem with the simple .travis.yml script is that it doesn't support ROS Kinetic, and I've been working hard to get the Kinetic code working so this is important.

The problem with the ``industrial_ci`` script are many:
- it is too [verbose and would timeout Travis builds](https://github.com/ros-industrial/industrial_ci/issues/66), so it was unusable for MoveIt!
- it has become really bloated with unnecessary features such as [abstracting catkin_tools](https://github.com/ros-industrial/industrial_ci/issues/72) and removing antiquated [manifest.xml](https://github.com/ros-industrial/industrial_ci/blob/master/rosdep-install.sh#L32) files
- still had code from [hydro](https://github.com/ros-industrial/industrial_ci/blob/master/travis.sh#L158)
- used a weird [folding style](https://github.com/ros-industrial/industrial_ci/issues/55) that was not as easy to read as regular travis files
- required [long build times in Docker](https://github.com/ros-industrial/industrial_ci/pull/52/files#r68967096)

It is certainly possible to fix these things in industrial_ci, as [I tried](https://github.com/ros-industrial/industrial_ci/pull/70) but the turn around time seemed kind of [slow](https://github.com/ros-planning/moveit_core/pull/304#issuecomment-229397314) because changing anything in this script could potentially break many other projects in the ROS Industrial project using ``industrial_ci``. @rhaschke came to the [same conclusion](https://github.com/ros-industrial/industrial_ci/pull/48#issuecomment-219294989)

> Thinking more about the issue, I came to the conclusion that it is hard to server the two goals of i) printing the commands from travis.sh but ii) not printing the commands in called functions or sourced scripts.
The output in the web interface is cluttered already in the overview (with travis_time_start, set +x, and <<< >>>. Hence, for MoveIt, I resumed to standard travis.yml files, improving them according to some inspiration gained from your script. I like the result much better - but that's my subjective view of course.

I've addressed his issue by making the console output [really pretty](https://travis-ci.org/ros-planning/moveit_core/builds/143428552). This new script uses [customized MoveIt! Docker containers](https://hub.docker.com/r/moveit/moveit_docker/builds/).

So I propose creating a ``moveit_ci`` repo and moving my https://github.com/davetcoleman/moveit_ci project there. See the README in that project for more details on the script.

This new script has [solved building Kinetic for MoveIt!](https://github.com/ros-planning/moveit_core/pull/304)

davetcoleman, 2016-07-08:
@130s I'm mostly looking for your feedback

rhaschke, 2016-07-08:
+1 Great job!

davetcoleman, 2016-07-14:
The two places we currently needed this new CI method has been merged, so this issue is resolved for now. It would be nice if all moveit_* repos used this new script, but I don't think its critical at this moment to do so. In fact, I'd rather all those places just got merged into a single repo (future topic). Closing this issue.

davetcoleman, 2016-07-19:
Closing this issue since there seems to be acceptance - @130s has started using this script in various places such as https://github.com/ros-interactive-manipulation/manipulation_msgs/pull/8

closed 2016-07-19 by davetcoleman