moveit/moveit_core #310: C++11 Support for all moveit_* repos

davetcoleman, 2016-07-14:
It's [come to my attention](https://github.com/davetcoleman/moveit_visual_tools/issues/9) from @simonschmeisser that there is a linking error from any package that wants to use geometric_shapes with C++11 because of boost::variant c++11/c++03 incompatibilities. Because kinetic minimum requirements includes C++11, I think we should upgrade geometric_shapes to C++11. However, this would also require use to upgrade moveit_core since it depends on geometric_shapes. Which in turn would require us to upgrade all of MoveIt!, I believe. This is necessary sooner or later anyway. Thoughts? We've already upgraded moveit_planners here https://github.com/ros-planning/moveit_planners/pull/73 because of OMPL

simonschmeisser, 2016-07-14:
fcl also moved to c++11 in fcl > 0.4.0 and thus adds a c++11 requirement on moveit_core. fcl > 0.4.0 is required as fcl <= 0.4.0 is not compatible to octomap 1.8 as released in kinetic

https://github.com/flexible-collision-library/fcl/issues/137

v4hn, 2016-07-14:
+1 just add the respective compiler option to the repos.

davetcoleman, 2016-07-15:
The three above PRs for C++11 are ready to be merged together (geometric_shapes, moveit_core, moveit_ros)

rhaschke, 2016-07-16:
@davetcoleman Go ahead and merge those PRs whenever you see fit.

davetcoleman, 2016-07-16:
I have merged the three PRs

closed 2016-07-16 by davetcoleman