moveit/moveit_core #313: Build from source instructions for Kinetic and FCL

davetcoleman, 2016-07-19:
This is a temporary issue until https://github.com/ros-planning/moveit_core/issues/299 is fixed for releasing debians

Currently the kinetic branches of MoveIt! build and pass their tests, but do not actually run

@simonschmeisser reported on the mailing list:

> I have been using kinetic for a while now, here is my rosinstall file: https://gist.github.com/simonschmeisser/3fd113bada14ae0a7ac159cfe8b861d7
> Basically I am currently using octomap 1.7 and fcl 0.4. That compiles and runs fine.

I was unable to reproduce his results just now using both his and [my rosinstall](https://gist.github.com/davetcoleman/8607010041b41767fbb261397d8f1425) and am instead getting API build errors:

```
Errors     << moveit_core:make /root/ws_moveit/logs/moveit_core/build.make.001.log
In file included from /root/ws_moveit/install/include/fcl/traversal/traversal_node_setup.h:48:0,
                 from /root/ws_moveit/src/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp:40:
/root/ws_moveit/install/include/fcl/traversal/traversal_node_octree.h: In member function ‘bool fcl::OcTreeSolver<NarrowPhaseSolver>::OcTreeShapeDistanceRecurse(const fcl::OcTree*, const OcTreeNode*, const fcl::AABB&, const S&, const fcl::AABB&, const fcl::Transform3f&, const fcl::Transform3f&) const’:
/root/ws_moveit/install/include/fcl/traversal/traversal_node_octree.h:268:50: error: ‘const OcTreeNode {aka const class octomap::OcTreeNode}’ has no member named ‘getChild’
         const OcTree::OcTreeNode* child = root1->getChild(i);
```

@simonschmeisser can you give more details to your setup?

I also would like to add for others: if you want to build FCL in a catkin workspace you first need to get its package.xml:

> wget https://raw.githubusercontent.com/ros-gbp/fcl-release/debian/jade/fcl/package.xml

Otherwise rosdep will still require you to install Ubuntu's version of libfcl-dev

anderwm, 2016-07-19:
This looks like cmake is still finding octomap 1.8 [previous issue](https://github.com/flexible-collision-library/fcl/issues/126)

simonschmeisser, 2016-07-21:
I have added a file src/octomap/package.xml:
```
<?xml version="1.0"?>
<package>
  <name>octomap</name>
  <version>1.7.2</version>
  <description>The foo package</description>

  <maintainer email="user@todo.todo">user</maintainer>
  <license>BSD</license>

  <buildtool_depend>cmake</buildtool_depend>

  <build_depend>boost</build_depend>

  <run_depend>boost</run_depend>
  <run_depend>catkin</run_depend>

  <export>
    <build_type>cmake</build_type>
  </export>
</package>

```

as well as src/fcl/package.xml
```

<?xml version="1.0"?>
<package>
  <name>fcl</name>
  <version>0.4.0</version>
  <description>The foo package</description>

  <maintainer email="user@todo.todo">user</maintainer>
  <license>BSD</license>

  <buildtool_depend>cmake</buildtool_depend>

  <build_depend>boost</build_depend>
  <build_depend>octomap</build_depend>

  <run_depend>boost</run_depend>
  <run_depend>catkin</run_depend>

  <export>
    <build_type>cmake</build_type>
  </export>
</package>

```

the build_depend fixes the dependency of fcl on octomap, leading to octomap being build first. it will then be preferred over the system octomap

davetcoleman, 2016-07-21:
Thanks @simonschmeisser for the package.xml tip.

So if we build ``fcl`` from source in that manner, I think we need to change the moveit_core package.xml fcl dependency to just ready ``fcl`` instead of ``libfcl-dev`` - does that sound right? Otherwise rosdep will try to install the system version.

I also had to install ``libqglviewer-dev-qt4`` through apt-get. I still got a moveit_core error with FCL:

```
make[2]: *** No rule to make target '/usr/lib/libfcl.so', needed by '/root/ws_moveit/devel/.private/moveit_core/lib/libmoveit_collision_detection_fcl.so'.  Stop.
```

I then tried ``catkin config --install`` and it fixed that problem but there still seems to be a major API change with FCL:

```
In file included from /root/ws_moveit/install/include/fcl/traversal/traversal_node_setup.h:48:0,
                 from /root/ws_moveit/src/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp:40:
/root/ws_moveit/install/include/fcl/traversal/traversal_node_octree.h: In member function ‘bool fcl::OcTreeSolver<NarrowPhaseSolver>::OcTreeShapeDistanceRecurse(const fcl::OcTree*, const OcTreeNode*, const fcl::AABB&, const S&, const fcl::AABB&, const fcl::Transform3f&, const fcl::Transform3f&) const’:
/root/ws_moveit/install/include/fcl/traversal/traversal_node_octree.h:268:50: error: ‘const OcTreeNode {aka const class octomap::OcTreeNode}’ has no member named ‘getChild’
         const OcTree::OcTreeNode* child = root1->getChild(i);
```

@simonschmeisser how did you fix this ``getChild(i)`` missing member function?

anderwm, 2016-07-21:
I'm not as good as I should be with cmake, but I still can't make it stop finding 1.8 even with your package.xml files.  I also tried to add version requirements to the moveit's find_package

davetcoleman, 2016-07-21:
``ros-kinetic-octomap`` is 1.7.0, so that is good. Is there a system (debian) version I don't know about? 
The octomap source version @simonschmeisser is recommending is 1.7.2  - should it be 1.7.0?
Does octomap's ``package.xml`` go in the root folder or the ``octomap/octomap`` folder? Didn't make a difference on my build.

anderwm, 2016-07-21:
Ok, now I'm really confused.  I thought all of these problems were stemming from the fact that ros-kinetic uses octomap 1.8 (I learned that from [here](http://wiki.ros.org/kinetic/Migration)).  After learning that, I tried to compile fcl with octomap 1.8 and failed just like we are failing now.  This led me to [octomap 1.8 release notes](https://github.com/OctoMap/octomap/releases/tag/v1.8.0), and to file [issue 126](https://github.com/flexible-collision-library/fcl/issues/126) with fcl.

It looks to me like @simonschmeisser is fixing all this by forcing octomap 1.7.x which allows everything to compile as it did in Jade.  But I'm still new to this world.

simonschmeisser, 2016-07-22:
Well fcl 0.5 is out since yesterday, so the real fix would be to upgrade to that. Make sure to double check that your git branches are actually correct. Octomap 1.7.2 is still compatible to 1.7.0, the changes were only introduced in 1.8

Can you confirm that octomap finishes building before fcl starts? Do a catkin clean --all to ensure that cmake isn't caching the path to system octomap

v4hn, 2016-07-22:
On Thu, Jul 21, 2016 at 01:52:26PM -0700, Dave Coleman wrote:
> ``ros-kinetic-octomap`` is 1.7.0, so that is good.

According to [rosdistro](https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml#L2250) it is not.
Did I miss something?

davetcoleman, 2016-07-22:
> According to [rosdistro](https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml#L2250) it is not.

Looks like my Docker images aren't up to date! They should auto-rebuild with the OSRF images are rebuilt, which I had checked and they seemed to rebuild often. Maybe I just have a local image that is out of date.

davetcoleman, 2016-07-25:
> Can you confirm that octomap finishes building before fcl starts?

Yes it does

> are you doing catkin_make_isolated since the cmake files in fcl/octomap don't call catkin_package()? Or did you change the cmake files in those packages?

I'm using ``catkin_tools``, which I highly recommend. It does the isolated build automatically. 

@anderwm I'm seeing some success with @de-vri-es new PRs, over [here](https://github.com/ros-planning/moveit_core/pull/316#issuecomment-235012510). I'm going to wait for the rest of his PRs to fix my compile problems

davetcoleman, 2016-07-27:
Building Kinetic from source now works - see https://github.com/davetcoleman/moveit_kinetic_cpp11

closed 2016-07-27 by davetcoleman